adafruit/Adafruit_CircuitPython_SI7021 #27: Heater control not implemented

aalhard, 2021-12-26:
the proccess in 5.2 of this https://www.silabs.com/documents/public/data-sheets/Si7021-A20.pdf

Arduino has it

caternuson, 2021-12-26:
Possible new property interface?
```python
# get heater state
si7021.heater_level
# set heater on
si7021.heater_level = 0  # on levels 0 (min) to 15 (max)
# set heater off
si7021.heaterl_level = None
```

rsbohn, 2021-12-26:
Looks like you can turn the heater on or off without changing the heater level.

caternuson, 2021-12-26:
Yep, it can. The suggestion above combines turning on/off and setting level in one step, with the goal of trying to be as simple as possible - i.e. it's one line of code.

Would separate on/off and level control be more useful? I think it would require two properties:
```python
si7021.heater_level = 5
si7021.heater_enabled = True
```

PR #29 (2022-03-02): Add heater control functionality
Resolves #27 by adding heater control in a single property which combines enabling and setting.  Uses `adafruit_register` so that dependency has been added.  Example usage in comment in the simpletest file.

**NOT TESTED** - Hardware isn't available for purchase right now so if anyone has it, please let me know!
closed 2022-03-02 by tekktrik