adafruit/Adafruit_CircuitPython_SI7021 #28: Cannot run example code twice on Raspberry Pi

jonbobcar, 2021-12-31:
After following the Learn guide to set the sensor up on a Raspberry Pi 3b and successfully running the full example code one time, running the example code any subsequent time results in the following error:

```
Traceback (most recent call last):
  File "/home/pi/si7021/environ.py", line 12, in <module>
    sensor = adafruit_si7021.SI7021(board.I2C())
  File "/usr/local/lib/python3.9/dist-packages/adafruit_si7021.py", line 139, in __init__
    raise RuntimeError("bad USER1 register (%x!=%x)" % (value, _USER1_VAL))
RuntimeError: bad USER1 register (0!=3a)
```

Rebooting the Pi and trying again results in the same error, but without the code executing correctly even one time.

`python --version`
`Python 3.9.2`

`cat /etc/os-release`
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

jonbobcar, 2021-12-31:
For posterity, here is the full example code:

```
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""
Initializes the sensor, gets and prints readings every two seconds.
"""
import time
import board
import adafruit_si7021

# Create library object using our Bus I2C port
sensor = adafruit_si7021.SI7021(board.I2C())

temp = []
humidity = []

while True:
    print("\nTemperature: %0.1f C" % sensor.temperature)
    print("Humidity: %0.1f %%" % sensor.relative_humidity)
    time.sleep(2)
```

jonbobcar, 2022-01-06:
I seem to have fixed the issue by adding the following to `/boot/config.txt`

```
# Clock stretching by slowing down to 10KHz
dtparam=i2c_arm_baudrate=10000
```

Found that solution here:

https://forums.adafruit.com/viewtopic.php?f=60&t=180463&p=878063&hilit=si7021#p878063

Perhaps this should be added to the documentation for the SI7021 breakout page?

caternuson, 2022-01-15:
Yeah clock stretching! The Si7021 can potentially clock stretch for a couple of reasons:
[image]

The driver is using the "No Hold" (no clock stretching) commands for making the measurements:
[image]
https://github.com/adafruit/Adafruit_CircuitPython_SI7021/blob/ad42ee292232a3371176bc115d7f791c6b9b0a7c/adafruit_si7021.py#L35-L36

Seems like it should work OK at 100kHz? But maybe not. Either way, yah, agree worth adding warning and info to learn guide for this. 
https://learn.adafruit.com/adafruit-si7021-temperature-plus-humidity-sensor/circuitpython-code

jonbobcar, 2022-01-15:
So this was two weeks ago and I've started and restarted this code a dozen or more times and it has been working great at 10 KHz.

To check 100 KHz I tried

```# Clock stretching by slowing down to 100KHz
dtparam=i2c_arm_baudrate=100000```

in `/boot/config.txt` which resulted in the same error.

Thanks for the confirmation that this is a good fix! I'll let you guys close it in case you want to remember to add a separate issue in regards to the documentation change.

caternuson, 2022-01-17:
Docs updated. Pretty simple, just needed to add a call out and link to existing general I2C fix page:
https://learn.adafruit.com/adafruit-si7021-temperature-plus-humidity-sensor/circuitpython-code#python-computer-wiring-2997620-5

closed 2022-01-17 by caternuson