adafruit/Adafruit_CircuitPython_SI7021 #31: Issues with Pypi installation and Blinka

stonehippo, 2022-03-13:
I'm having issues getting the latest release of this module installed via Pypi. For example, `sudo pip install adafruit_circuitpython_si7021` appears to work, but there's no implementation installed.

The latest release showing on Pypi is 3.4.0, not 4.0.0. Looking at the assets in the [4.0.0 release](https://github.com/adafruit/Adafruit_CircuitPython_SI7021/releases/tag/4.0.0), it looks like everything still refers to 3.4.0. And when I check my `dist-packages` directory, the dist-info for the package is there, but no implementation files. I've cleared out my pip cache, and tried this on a couple of systems, but it definitely seems like things are broken in Pypi.

I've also confirmed that the package is fine in the Circuitpython library bundle and can be installed correctly via copying or `circup`.

tekktrik, 2022-03-13:
Sorry, there was a known issue with the last release being a rerelease to 4.0.0, but since it's identical to 3.4.0, PyPI got angry and wouldn't accept it.  Release of #30 will provide the opportunity to push another release to PyPI, and the two will sync back up!

stonehippo, 2022-03-13:
@tekktrik ah, got it!

closed 2022-03-14 by FoamyGuy