adafruit/Adafruit_CircuitPython_SI7021 #32: bad USER1 register with Blinka

stonehippo, 2022-03-23:
I've got an si7021 hooked up to I2C on a RPi 3A with Blinka, and I'm an error when initializing the module. The code couldn't be simpler:

```python
import board
import adafruit_si7021

si = adafruit_si7021.SI7021(board.I2C())
```

This works fine in "native" CircuitPython on something like a RP2040 Feather, but is failing with the following on the RPI/Blinka:

> Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/adafruit_si7021/__init__.py", line 153, in __init__
    raise RuntimeError("bad USER1 register (%x!=%x)" % (value, _USER1_VAL))
RuntimeError: bad USER1 register (0!=3a)

I might take a peek at the I2C comms with my I2CDriver to see what's going on, since the code in question seems to be a pretty simple write the read. And I know that it works elsewhere.

stonehippo, 2022-03-23:
Ok, it's a timing issue. I added `time.sleep(.25)` after the reset and there's no issue. I suspect that this is a problem related to the clock-stretching bug on the RPi I2C bus.

caternuson, 2022-03-23:
Do you have the Pi's I2C bus clock slowed down to deal with clock stretching?

stonehippo, 2022-03-23:
I just knocked the Pi bus clock down to 10K on this device and things work great, so it's almost certainly the clock stretching bug.

caternuson, 2022-03-23:
It's a known "feature". The slow down is recommended in the guide:
https://learn.adafruit.com/adafruit-si7021-temperature-plus-humidity-sensor/circuitpython-code#python-computer-wiring-2997620-5

stonehippo, 2022-03-23:
Thanks, I missed that it was called out in the guide. Closing this one

closed 2022-03-23 by stonehippo