adafruit/Adafruit_CircuitPython_SI7021 #35: Raises RuntimeError with Raspberry Pi / Blinka

FoamyGuy, 2023-04-17:
```
Traceback (most recent call last):
  File "/home/timc/Adafruit_CircuitPython_SI7021/si7021_simpletest.py", line 14, in <module>
    sensor = adafruit_si7021.SI7021(i2c)
  File "/home/timc/Adafruit_CircuitPython_SI7021/adafruit_si7021/__init__.py", line 153, in __init__
    raise RuntimeError("bad USER1 register (%x!=%x)" % (value, _USER1_VAL))
RuntimeError: bad USER1 register (0!=3a)
```
Followed wiring and setup from here: https://learn.adafruit.com/adafruit-si7021-temperature-plus-humidity-sensor/circuitpython-code#python-computer-wiring-2997620

I2C Scan does show that there is a device with address 0x40 connected. 

Initializing the library always fails with the above error. 

I tested the same breakout on a Feather S2 microcontroller with the library and it does work in that context. This issue seems specific to RasPi / Blinka.

FoamyGuy, 2023-04-17:
It's only been a handful of trials, and I changed some other things in addition so I need to go back and isolate down to a single change. But I believe this issue might be caused by not waiting long enough after the RESET command is issued here: https://github.com/adafruit/Adafruit_CircuitPython_SI7021/blob/318333e3a13b7abc35d3fc677ed9eea4e0d2e39e/adafruit_si7021/__init__.py#L139

In the current code, right after that command is issued a loop starts up trying to read the USER1 register and if the value it finds there isn't what is expected it raises the exception mentioned in the issue post. 

I added a small `time.sleep(0.1)` after the RESET command and it the simpletest seems to run reliably now with that sleep included. Perhaps a smaller sleep would work as well, I haven't tested other values, and I did actually eventually a few successful runs even without the sleep (meaning there is some variability in how fast is _too fast_)

KeithTheEE, 2023-04-17:
Is the `time.sleep(0.1)` only necessary during the init, and no where else?

FoamyGuy, 2023-04-17:
@KeithTheEE seems like yes. I added it directly after the RESET before entering the loop and it seems to work consistently that way. 

The time can actually be even smaller as well. I had success with `0.01` and even `0.001`. Still need to do more thorough testing because I only ran it a few times with the smaller values.

KeithTheEE, 2023-04-17:
Awesome. I'm guessing setting `time.sleep` to 0.015 will work consistently. Based off of the [datasheet's](https://cdn-learn.adafruit.com/assets/assets/000/035/931/original/Support_Documents_TechnicalDocs_Si7021-A20.pdf) "Table 2. General Specifications (Continued)" under the "Powerup Time", "After issuing a software reset command" (middle of page 5). I *think* that is the condition you're running into, but I could be reading the datasheet wrong. It took me ages to find that timing information and there could be another table somewhere that actually applies here.

PR #36 (2023-04-17): sleep after reset before read register
resolves #35 

It turns out a sleep of only `0.015` should be the longest that is needed, as specified in the datasheet (Thank you to @KeithTheEE  and @jposada202020 !)

I tested the change successfully on Raspberry Pi 3 B+ w/ Blinka and double checked it on a Feather S2 TFT to ensure the microcontroller behavior still works as expected.
closed 2023-04-17 by FoamyGuy